TerosTechnology/colibri #27: run.py

qarlosalberto, 2020-03-17:
- pre_check()
- post_check()
- Synopsys libraries
- Disable ieee warnings
- New run.py sytle:
https://github.com/TerosTechnology/TerosHDLbackend/tree/beautifuler
**beautifuler branch**

qarlosalberto, 2020-05-10:
{
  checks : true/false,
  ghdl_support : [
    synopsys_libraries : true/false,
    disable_ieee_warnings: true/false,
    code_coverage : true/false
  ]
  modelsim_support : [
    gui : true/false
  ]
}

smgl9, 2020-05-10:
Proposal for the new interface:

```
{
  "src":["./exampleRunpy_1.vhd"],
  "tb": ["./tbVhdlVunitRunpy.vhd"],
  "lang": ["vhdl"],
  "config": { "name": "test_1", "outputPath": "./", "codeCovPath": "covHtml","checks": true },
  "simulator": ["ghdl/modelsim"],
  "ghdl_support":{"synopsys_libraries" : true, "disable_ieee_warnings": true, "code_coverage": true},
  "modelsim_support":{"gui" : true},
  "libraries": { "xilib_path": "xilib_path"}
}
```

qarlosalberto, 2020-05-10:
```
{
  "src": [
    "./exampleRunpy_1.vhd"
  ],
  "tb": [
    "./tbVhdlVunitRunpy.vhd"
  ],
  "lang": "vhdl",
  "config": {
    "name": "test_1",
    "checks": true,
    "simulator_suport": {
      "ghdl" : {
        "enable" : true,
        "config" : {
          "synopsys_libraries" : true,
          "disable_ieee_warnings" : true,
          "code_coverage" : {
            "enable" : true,
            "output_path": "/output/path"
          }
        }
      },
      "modelsim" : {
        "enable" : true
      }
    }
  }
}
```

smgl9, 2020-05-10:
output path means outputfor runpy. Because of that, new proposal:
```
{
  "src": [
    "./exampleRunpy_1.vhd"
  ],
  "tb": [
    "./tbVhdlVunitRunpy.vhd"
  ],
  "lang": "vhdl",
  "config": {
    "name": "test_1",
    "output_path": "/output/path"
    "checks": true,
    "simulator_suport": {
      "ghdl" : {
        "enable" : true,
        "config" : {
          "synopsys_libraries" : true,
          "disable_ieee_warnings" : true,
          "code_coverage" : {
            "enable" : true
          }
        }
      },
      "modelsim" : {
        "enable" : true
      }
    }
  }
}
```

closed 2020-05-24 by qarlosalberto