TerosTechnology/colibri #42: Bug VHDL parser Windows

smgl9, 2020-03-22:


smgl9, 2020-03-22:
The paser fails on Windows systems

[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.45.0 x64
**Electron**: 4.2.7
**OS**: Unknown Windows version
**Thrown From**: [TerosHDL](https://github.com/TerosTechnology/terosHDL) package 1.1.0

### Stack Trace

Uncaught Error: Command failed: python3 C:\Users\isma\Downloads\terosHDL-colibri\terosHDL-colibri\node_modules\jsteros\src\parser\parser.py ""  "--------------------------------------------------------------
-- GCD design using FSMD (ESD book Figure 2.9)
-- Weijun Zhang, 04/2001
--
-- GCD algorithm behavior modeling (GCD.vhd)
-- the calculator has two 4-bit inputs and one output
--
-- NOTE: idle state required to obtain
-- the correct synthesis results
--------------------------------------------------------------

library ieee;
use ieee.std_logic_1164.all;
use ieee.std_logic_arith.all;
use work.all;

--------------------------------------------------------------

entity gcd is

port(	clk: 	in std_logic;
	rst:	in std_logic;
	go_i:	in std_logic;
	x_i:	in unsigned(3 downto 0);
	y_i:	in unsigned(3 downto 0);
	d_o:	out unsigned(3 downto 0)
);
end gcd;

--------------------------------------------------------------

architecture FSMD of gcd is
begin

    fsdmp:process(rst, clk)

    -- define states using variable
    type S_Type is (ST0, ST1, ST2);
    variable State: S_Type := ST0 ;
    variable Data_X, Data_Y: unsigned(3 downto 0);

    begin

        if (rst='1') then		    -- initialization
	    d_o <= \"0000\";
	    State := ST0;
	elsif (clk'event and clk='1') then
	    case State is
	        when ST0 =>		    -- starting
		    if (go_i='1') then
			Data_X := x_i;
			Data_Y := y_i;
			State := ST1;
		    else
		        State := ST0;
		    end if;
[... 122 lines cut ...]
    at EventEmitter.t (~/AppData/Local/atom/app-1.45.0/resources/app/static/<embedded>:1:760343)
    at EventEmitter.emit (events.js:182:13)
```

### Commands

```
     -8:35.1.0 TerosHDL:copy-as-component (span.syntax--meta.syntax--list.syntax--element.syntax--vhdl)
     -8:25.7.0 TerosHDL:copy-as-tb-cocotb (span.syntax--meta.syntax--list.syntax--element.syntax--vhdl)
  6x -7:49.4.0 application:open-file (a.btn-copy-report.icon.icon-clippy)
     -6:47.6.0 TerosHDL:copy-as-image (input.hidden-input)
  3x -6:34.5.0 application:open-file (input.hidden-input)
     -6:01.3.0 TerosHDL:copy-as-image (input.hidden-input)
     -0:12.4.0 TerosHDL:copy-as-component (span.syntax--meta.syntax--list.syntax--element.syntax--vhdl)
```

### Non-Core Packages

```
console-panel 0.3.1 
language-vhdl 0.7.2 
linter 2.3.1 
TerosHDL 1.1.0 
tool-bar 1.2.2 
```

qarlosalberto, 2020-03-22:
```{bash}
python3 C:\Users\isma\Downloads\terosHDL-colibri\terosHDL-colibri\node_modules\jsteros\src\parser\parser.py ""  "--------------------------------------------------------------
-- GCD design using FSMD (ESD book Figure 2.9)
-- Weijun Zhang, 04/2001
--
-- GCD algorithm behavior modeling (GCD.vhd)
-- the calculator has two 4-bit inputs and one output
--
-- NOTE: idle state required to obtain
-- the correct synthesis results
--------------------------------------------------------------

library ieee;
use ieee.std_logic_1164.all;
use ieee.std_logic_arith.all;
use work.all;

--------------------------------------------------------------

entity gcd is

port(	clk: 	in std_logic;
	rst:	in std_logic;
	go_i:	in std_logic;
	x_i:	in unsigned(3 downto 0);
	y_i:	in unsigned(3 downto 0);
	d_o:	out unsigned(3 downto 0)
);
end gcd;

--------------------------------------------------------------

architecture FSMD of gcd is
begin

    fsdmp:process(rst, clk)

    -- define states using variable
    type S_Type is (ST0, ST1, ST2);
    variable State: S_Type := ST0 ;
    variable Data_X, Data_Y: unsigned(3 downto 0);

    begin

        if (rst='1') then		    -- initialization
	    d_o <= \"0000\";
	    State := ST0;
	elsif (clk'event and clk='1') then
	    case State is
	        when ST0 =>		    -- starting
		    if (go_i='1') then
			Data_X := x_i;
			Data_Y := y_i;
			State := ST1;
		    else
		        State := ST0;
		    end if;
		when ST1 =>		    -- idle state
		    State := ST2;
		when ST2 =>		    -- computation
		    if (Data_X/=Data_Y) then
		        if (Data_X<Data_Y) then
			    Data_Y := Data_Y - Data_X;
			else
			    Data_X := Data_X - Data_Y;
			end if;
			State := ST1;
		    else
			d_o <=Data_X;       -- done
		        State := ST0;
		    end if;
		when others =>		    -- go back
		    d_o <= \"ZZZZ\";
		    State := ST0;
            end case;
 	end if;

    end process;

end FSMD;

--------------------------------------------------------------
 "

```
At child_process.js:661

Error: Command failed: python3 C:\Users\isma\Downloads\terosHDL-colibri\terosHDL-colibri\node_modules\jsteros\src\parser\parser.py ""  "--------------------------------------------------------------
-- GCD design using FSMD (ESD book Figure 2.9)
-- Weijun Zhang, 04/2001
--
-- GCD algorithm behavior modeling (GCD.vhd)
-- the calculator has two 4-bit inputs and one output
--
-- NOTE: idle state required to obtain
-- the correct synthesis results
--------------------------------------------------------------

library ieee;
use ieee.std_logic_1164.all;
use ieee.std_logic_arith.all;
use work.all;

--------------------------------------------------------------

entity gcd is

port(	clk: 	in std_logic;
	rst:	in std_logic;
	go_i:	in std_logic;
	x_i:	in unsigned(3 downto 0);
	y_i:	in unsigned(3 downto 0);
	d_o:	out unsigned(3 downto 0)
);
end gcd;

--------------------------------------------------------------

architecture FSMD of gcd is
begin

    fsdmp:process(rst, clk)

    -- define states using variable
    type S_Type is (ST0, ST1, ST2);
    variable State: S_Type := ST0 ;
    variable Data_X, Data_Y: unsigned(3 downto 0);

    begin

        if (rst='1') then		    -- initialization
	    d_o <= \"0000\";
	    State := ST0;
	elsif (clk'event and clk='1') then
	    case State is
	        when ST0 =>		    -- starting
		    if (go_i='1') then
			Data_X := x_i;
			Data_Y := y_i;
			State := ST1;
		    else
		        State := ST0;
		    end if;
		when ST1 =>		    -- idle state
		    State := ST2;
		when ST2 =>		    -- computation
		    if (Data_X/=Data_Y) then
		        if (Data_X<Data_Y) then
			    Data_Y := Data_Y - Data_X;
			else
			    Data_X := Data_X - Data_Y;
			end if;
			State := ST1;
		    else
			d_o <=Data_X;       -- done
		        State := ST0;
		    end if;
		when others =>		    -- go back
		    d_o <= \"ZZZZ\";
		    State := ST0;
            end case;
 	end if;

    end process;

end FSMD;

--------------------------------------------------------------
 "

```

qarlosalberto, 2020-10-04:
Could you check if it has been fixed?

smgl9, 2020-10-05:
Yes! Everything seems to be working fine now

closed 2020-10-05 by smgl9