TerosTechnology/colibri #74: fix template test

smgl9, 2020-05-24:


smgl9, 2020-05-26:
@qarlosalberto 

I'm not sure about the use of runpy template. It seems that the configuration is been parsed like code. And i guess that it shouldn't. Can you take a look and tell me what you think?

qarlosalberto, 2020-05-27:
We have to change the template interface and to add the runpy engine.

1. In the constructor only should pass the type and a dicctionary with the language: 

async getTemplate(type, {language:"vhdl"}){ }

2. Pass the code and the options in:

generate(code, {option_1: "", option_2: ""}){}

3. The linter will be instance in the generate function.

smgl9, 2020-06-03:
Interface proposal:

getTemplate(type, {language:"vhdl"}){ }
async generate(code, {option_1: "", option_2: ""}){}

- "options" are optional.
- "language" is part of the template constructor.
- Maybe "language" integrated in "type" ¿?

qarlosalberto, 2020-06-04:
It's ok. I think that generate() should be different in each template. For example:

- run.py --> generate(options)
- components --> generate(code, type, options)

And language shouldn't be inside "options", because it's not an optional argument. It's enough?:

get_template(type,language)

Or do you need the options parameter in the constructor of some template? Language can be defined (in component, testbench... etc) and undefined in run.py

qarlosalberto, 2020-06-04:
Also, you need to check (try-catch) when you call the parser. There may be an error in the parser, and the template generator musn't crash.

smgl9, 2020-06-05:
That's the idea. And we do not need "options" for any constructor