TerosTechnology/colibri #86: Problems with "run test" in Linux

tpcorrea, 2020-06-05:
My system is Ubuntu 18.04.4, python 2.7, VUnit 4.3 and ghdl 0.37.

When I try to run a simulation through TerosHDL menu, I get the error:

> Compiling into vunit_lib: ../../../../../.local/lib/python2.7/site-packages/vunit/vhdl/data_types/src/codec.vhd                             
> 
> failed
> === Command used: ===
> /usr/local/bin/ghdl -a --workdir=/home/tomaspcorrea/Repo/ssvect_redboard/ip_repo/ssvectors_ctrl_1.0/vunit/vunit_out/ghdl/libraries/vunit_lib --work=vunit_lib --std=08 -P/home/tomaspcorrea/Repo/ssvect_redboard/ip_repo/ssvectors_ctrl_1.0/vunit/vunit_out/ghdl/libraries/vunit_lib -P/home/tomaspcorrea/lib/ghdl/xilinx-vivado/unisim/v08 -P/home/tomaspcorrea/lib/ghdl/xilinx-vivado/unifast/v08 -P/home/tomaspcorrea/lib/ghdl/xilinx-vivado/unimacro/v08 -P/home/tomaspcorrea/lib/ghdl/xilinx-vivado/secureip/v08 -P/home/tomaspcorrea/lib/ghdl/uvvm/uvvm_util/v08 -P/home/tomaspcorrea/lib/ghdl/uvvm/bitvis_vip_axilite/v08 -P/home/tomaspcorrea/Repo/ssvect_redboard/ip_repo/ssvectors_ctrl_1.0/vunit/vunit_out/ghdl/libraries/src_lib -P/home/tomaspcorrea/Repo/ssvect_redboard/ip_repo/ssvectors_ctrl_1.0/vunit/vunit_out/ghdl/libraries/tb_lib /home/tomaspcorrea/.local/lib/python2.7/site-packages/vunit/vhdl/data_types/src/codec.vhd
> 
> === Command output: ===
> /home/tomaspcorrea/.local/lib/python2.7/site-packages/vunit/vhdl/data_types/src/codec.vhd:520:47: prefix of array attribute must be an object name
> /home/tomaspcorrea/.local/lib/python2.7/site-packages/vunit/vhdl/data_types/src/codec.vhd:520:47:note: (you can use -frelaxed to turn this error into a warning)
> /home/tomaspcorrea/.local/lib/python2.7/site-packages/vunit/vhdl/data_types/src/codec.vhd:545:51: prefix of array attribute must be an object name
> /home/tomaspcorrea/.local/lib/python2.7/site-packages/vunit/vhdl/data_types/src/codec.vhd:563:58: prefix of array attribute must be an object name
> /home/tomaspcorrea/.local/lib/python2.7/site-packages/vunit/vhdl/data_types/src/codec.vhd:617:66: prefix of array attribute must be an object name
> /home/tomaspcorrea/.local/lib/python2.7/site-packages/vunit/vhdl/data_types/src/codec.vhd:635:64: prefix of array attribute must be an object name
> /home/tomaspcorrea/.local/lib/python2.7/site-packages/vunit/vhdl/data_types/src/codec.vhd:653:66: prefix of array attribute must be an object name
> /home/tomaspcorrea/.local/lib/python2.7/site-packages/vunit/vhdl/data_types/src/codec.vhd:671:64: prefix of array attribute must be an object name
> 
> Compile failed

Searching for this issue, I found a bug report on vunit repo that was fixed in version 4.4.0 (see https://github.com/VUnit/vunit/pull/631). The problem is that the vunit version 4.4.0 can only be installed for python 3 and TerosHDL is calling "python", that corresponds to python 2.7 in Linux.

qarlosalberto, 2020-06-05:
Can you try to install TerosHDL manually in the branch 68-error-python3? It should fix the problem. Now you can configure the python in the menu.

For a manual installation:
```
apm install
```

smgl9, 2020-06-05:
https://github.com/TerosTechnology/terosHDL/tree/68-error-python3

tpcorrea, 2020-06-06:
Hi, how is the procedure? I tried to clone from the repo, switch to the
branch 68-error-python3 and run apm install inside the local repo.

KR,
Tomas

tpcorrea, 2020-06-06:
I have to clone it into the .atom/package folder, I just learned it. Well, when I try to run the test I just get in the console:

> /bin/sh: 1: undefined: not found

Before cloning and installing, I removed the previous terosHDL package.

smgl9, 2020-06-06:
@tpcorrea  Try to reproduce these steps, should be enough:

```
git clone -b 68-error-python3 https://github.com/TerosTechnology/terosHDL.git
cd terosHDL/
apm link
apm install
```
After the installation you have to reboot Atom.
If some errors come up try to undo the changes in the ./atom folder.
Hope it helps.

tpcorrea, 2020-06-06:
Hi, Ismael, yes I managed to install the 68-error-python3 branch, but then the run simulation buttom does not work. I get in the console:

> /bin/sh: 1: undefined: not found

When I switch to the master branch, I come back to the original problem with the codec.vhd file (described in the first post of this issue).

KR,
Tomas

qarlosalberto, 2020-06-06:
Can you try again? There was a typo error... :/

tpcorrea, 2020-06-06:
Carlos and Ismael,

seems to be fine. I still have issues in the compilation (I get the warning: package "axilite_bfm_pkg" is obsoleted by context "uvvm_util_context"), but this issue with the python version is ok.

Cheers,
Tomas

closed 2020-06-06 by tpcorrea

qarlosalberto, 2020-06-06:
Mm... Try to remove the folder vunit_out or push the clean button. You have compiled the libreries with python2 and python3, it causes conflicts