TerosTechnology/colibri #158: always documentation

smgl9, 2020-12-08:


qarlosalberto, 2020-12-08:
{
  entity: {
    name: "state_machine",
    comment: "",
  },
  generics: [
  ],
  ports: [
    {
      name: "clk",
      direction: "in",
      type: "STD_LOGIC",
      comment: "",
    },
    {
      name: "input",
      direction: "in",
      type: "STD_LOGIC",
      comment: "",
    },
    {
      name: "reset",
      direction: "in",
      type: "STD_LOGIC",
      comment: "",
    },
    {
      name: "output",
      direction: "out",
      type: "STD_LOGIC_VECTOR(1 downto 0)",
      comment: "",
    },
  ],
  body: {
    processes: [
      {
        name: "unnamed",
        description: "good comment 1\n",
      },
      {
        name: "stm_0",
        description: "",
      },
      {
        name: "unnamed",
        description: "good comment 55\ngood comment 4\n",
      },
      {
        name: "unnamed",
        description: "",
      },
      {
        name: "p_comb",
        description: "",
      },
    ],
    instantiations: [
      {
        name: "dut_instance",
        type: "work.dut(RTL)",
        description: "good comment 0\n",
      },
      {
        name: "HA",
        type: "HalfAdder",
        description: "",
      },
    ],
  },
  declarations: {
    types: [
      {
        name: "STATE_TYPE",
        type: "(s0, s1, s2)",
        line: 15,
        description: "type comment\n",
      },
      {
        name: "PEPE_TYPE",
        type: "(s0, s1, s2)",
        line: 17,
        description: "state 1 comment\n",
      },
    ],
    signals: [
      {
        name: "state",
        type: "STATE_TYPE",
        line: 19,
        description: "comment signal\n",
      },
      {
        name: "state1, state2, state3",
        type: "STATE_TYPE",
        line: 20,
        description: "prueba\n",
      },
      {
        name: "sample",
        type: "integer",
        line: 21,
        description: "prueba\n",
      },
    ],
    constants: [
      {
        name: "sample_cnt",
        type: "integer",
        default_value: "9",
        line: 22,
        description: "",
      },
    ],
  },
}

qarlosalberto, 2020-12-08:
```
{
  entity: {
    name: "state_machine",
    comment: "",
  },
  generics: [
  ],
  ports: [
    {
      name: "clk",
      direction: "in",
      type: "STD_LOGIC",
      comment: "",
    },
    {
      name: "input",
      direction: "in",
      type: "STD_LOGIC",
      comment: "",
    },
    {
      name: "reset",
      direction: "in",
      type: "STD_LOGIC",
      comment: "",
    },
    {
      name: "output",
      direction: "out",
      type: "STD_LOGIC_VECTOR(1 downto 0)",
      comment: "",
    },
  ],
  body: {
    processes: [
      {
        name: "unnamed",
        description: "good comment 1\n",
      },
      {
        name: "stm_0",
        description: "",
      },
      {
        name: "unnamed",
        description: "good comment 55\ngood comment 4\n",
      },
      {
        name: "unnamed",
        description: "",
      },
      {
        name: "p_comb",
        description: "",
      },
    ],
    instantiations: [
      {
        name: "dut_instance",
        type: "work.dut(RTL)",
        description: "good comment 0\n",
      },
      {
        name: "HA",
        type: "HalfAdder",
        description: "",
      },
    ],
  },
  declarations: {
    types: [
      {
        name: "STATE_TYPE",
        type: "(s0, s1, s2)",
        line: 15,
        description: "type comment\n",
      },
      {
        name: "PEPE_TYPE",
        type: "(s0, s1, s2)",
        line: 17,
        description: "state 1 comment\n",
      },
    ],
    signals: [
      {
        name: "state",
        type: "STATE_TYPE",
        line: 19,
        description: "comment signal\n",
      },
      {
        name: "state1, state2, state3",
        type: "STATE_TYPE",
        line: 20,
        description: "prueba\n",
      },
      {
        name: "sample",
        type: "integer",
        line: 21,
        description: "prueba\n",
      },
    ],
    constants: [
      {
        name: "sample_cnt",
        type: "integer",
        default_value: "9",
        line: 22,
        description: "",
      },
    ],
  },
}
```

qarlosalberto, 2020-12-08:
I think that it's better to change **comment** for **description**

smgl9, 2020-12-09:
It's ok. I think we can also include functions. In the same way that process

qarlosalberto, 2020-12-09:
Yes, and package support.

qarlosalberto, 2020-12-11:
I have added the field "default_value" to the generics.

qarlosalberto, 2020-12-11:
```
{
  entity: {
    name: "state_machine",
    description: " description\n",
  },
  generics: [
    {
      name: "SAMPLE",
      type: "integer",
      direction: "",
      default_value: "9",
      description: " comment multiline",
    },
    {
      name: "sample1",
      type: "std_logic",
      direction: "",
      default_value: "\"hola\"",
      description: "",
    },
    {
      name: "sample2",
      type: "integer",
      direction: "",
      default_value: "",
      description: "pepe",
    },
    {
      name: "sample3",
      type: "integer",
      direction: "",
      default_value: "",
      description: "pepe",
    },
    {
      name: "sample4",
      type: "integer",
      direction: "",
      default_value: "",
      description: "pepe",
    },
    {
      name: "output",
      type: "std_logic(1 downto 0)",
      direction: "",
      default_value: "",
      description: "",
    },
  ],
  ports: [
    {
      name: "clk",
      type: "STD_LOGIC",
      direction: "in",
      default_value: "",
      description: "",
    },
    {
      name: "input",
      type: "STD_LOGIC",
      direction: "in",
      default_value: "",
      description: "",
    },
    {
      name: "reset",
      type: "STD_LOGIC",
      direction: "in",
      default_value: "",
      description: " multiline comment",
    },
    {
      name: "output",
      type: "STD_LOGIC_VECTOR(1 downto 0)",
      direction: "out",
      default_value: "",
      description: " multiline comment",
    },
  ],
  body: {
    processes: [
      {
        name: "stm_0",
        description: "",
      },
      {
        name: "unnamed",
        description: " good comment 4\n",
      },
      {
        name: "unnamed",
        description: "",
      },
      {
        name: "p_comb",
        description: "",
      },
    ],
    instantiations: [
      {
        name: "dut_instance",
        type: "work.dut(RTL)",
        description: " good comment 0\n",
      },
      {
        name: "HA",
        type: "HalfAdder",
        description: "",
      },
    ],
  },
  declarations: {
    types: [
      {
        name: "STATE_TYPE",
        type: "(s0, s1, s2)",
        description: " type comment",
      },
      {
        name: "PEPE_TYPE",
        type: "(s0, s1, s2)",
        description: " state 1 comment",
      },
    ],
    signals: [
      {
        name: "state",
        type: "STATE_TYPE",
        description: " comment signal",
      },
      {
        name: "state1, state2, state3",
        type: "STATE_TYPE",
        description: " prueba",
      },
      {
        name: "sample",
        type: "integer",
        description: " prueba",
      },
    ],
    constants: [
      {
        name: "sample_cnt",
        type: "integer",
        default_value: "9",
        description: "",
      },
    ],
    functions: [
      {
        name: "CounterVal",
        description: "",
      },
    ],
  },
}
```

smgl9, 2020-12-11:
Ok. I will take it into account for verilog.

```
      {
        name: "state1, state2, state3",
        type: "STATE_TYPE",
        description: " prueba",
      }
```
If we want an unified criteria along with ports, we should split the signals in three different signals, in order to not to have a name with three signals.

qarlosalberto, 2020-12-11:
okk

smgl9, 2020-12-13:
Adding sensitivity list to processes

```
{
  entity: {
    name: "state_machine",
    description: " description\n",
  },
  generics: [
    {
      name: "SAMPLE",
      type: "integer",
      direction: "",
      default_value: "9",
      description: " comment multiline",
    },
    {
      name: "sample1",
      type: "std_logic",
      direction: "",
      default_value: "\"hola\"",
      description: "",
    },
    {
      name: "sample2",
      type: "integer",
      direction: "",
      default_value: "",
      description: "pepe",
    },
    {
      name: "sample3",
      type: "integer",
      direction: "",
      default_value: "",
      description: "pepe",
    },
    {
      name: "sample4",
      type: "integer",
      direction: "",
      default_value: "",
      description: "pepe",
    },
    {
      name: "output",
      type: "std_logic(1 downto 0)",
      direction: "",
      default_value: "",
      description: "",
    },
  ],
  ports: [
    {
      name: "clk",
      type: "STD_LOGIC",
      direction: "in",
      default_value: "",
      description: "",
    },
    {
      name: "input",
      type: "STD_LOGIC",
      direction: "in",
      default_value: "",
      description: "",
    },
    {
      name: "reset",
      type: "STD_LOGIC",
      direction: "in",
      default_value: "",
      description: " multiline comment",
    },
    {
      name: "output",
      type: "STD_LOGIC_VECTOR(1 downto 0)",
      direction: "out",
      default_value: "",
      description: " multiline comment",
    },
  ],
  body: {
    processes: [
      {
        name: "stm_0",
        sens_list: "@(posedge clk)", 
        description: "",
      },
      {
        name: "unnamed",
        sens_list: "@(posedge clk)",
        description: " good comment 4\n",
      },
      {
        name: "unnamed",
        sens_list: "@(posedge clk)",
        description: "",
      },
      {
        name: "p_comb",
        sens_list: "@(posedge clk)",
        description: "",
      },
    ],
    instantiations: [
      {
        name: "dut_instance",
        type: "work.dut(RTL)",
        description: " good comment 0\n",
      },
      {
        name: "HA",
        type: "HalfAdder",
        description: "",
      },
    ],
  },
  declarations: {
    types: [
      {
        name: "STATE_TYPE",
        type: "(s0, s1, s2)",
        description: " type comment",
      },
      {
        name: "PEPE_TYPE",
        type: "(s0, s1, s2)",
        description: " state 1 comment",
      },
    ],
    signals: [
      {
        name: "state",
        type: "STATE_TYPE",
        description: " comment signal",
      },
      {
        name: "state1, state2, state3",
        type: "STATE_TYPE",
        description: " prueba",
      },
      {
        name: "sample",
        type: "integer",
        description: " prueba",
      },
    ],
    constants: [
      {
        name: "sample_cnt",
        type: "integer",
        default_value: "9",
        description: "",
      },
    ],
    functions: [
      {
        name: "CounterVal",
        description: "",
      },
    ],
  },
}

closed 2020-12-26 by smgl9