TerosTechnology/colibri #160: Port name detection error for buses.

manub23, 2020-12-12:
Ports declared as 

[image] 

render as 

[image]

in the documentation.

smgl9, 2020-12-13:
Thank you for the feedback!

That was a bug in older versions but it is supposed to have been solved in the last release.
Can you confirm that you are using TerosHDL 0.0.8?
Have you installed the plugin from the vscode marketplace?

manub23, 2020-12-14:
@smgl9 Yup. I've tried TerosHDL 0.0.8 on VS Code on both Windows and Linux machines, only to get the same result.

smgl9, 2020-12-14:
Ok, 

Could you provide us a piece of code (omitting confidential/unnecessary parts) that is failing you? Because we have some test working fine with the same case of use of your screenshot

manub23, 2020-12-15:
Please find the code [here](https://gist.github.com/manub23/a26d0ccefc9bdd21d1c3bcb3e8b019f3).

smgl9, 2020-12-15:
I found the bug. It will be solved for the next release.
Thank you!

manub23, 2020-12-15:
Thank you, @smgl9

closed 2020-12-15 by manub23