TerosTechnology/colibri #175: Better "Than" Doxygen Support

el3ctrician, 2021-01-12:
First of all i would like to thank all of you guys for the awesome project. I think that, at least me, desperately needed some modern tools for Digital Designer as my older colleges tend to used notepad++ or vim or something else. You really did a great job so far and i would love to help see it improves and get better overtime. Following are some of my thoughts regarding the documentation tool "Documenter" that could be the best opensource tool available in my opinion. What i love most about it that it is clutter free as it gives useful information that could guide someone who reads the code for the first time while keeping the developer effort to minimum. In the past i was relying on Doxygen to get the job done but Doxygen generates a lot of useless stuff and repeated documentation when a small but efficient documentation is needed. Sure thing that it can be customized but it takes huge effort to do so specially for VHDL due to the lack of dedicated VHDL "mentality" since it treats VHDL some times as a generic OOP which is completly a mess.   

That said some of the features can be "borrowed" from Doxygen structure while adding support for new specific structure for digital design. 

For instance the command concept of Doxygen, more infos [here](https://www.doxygen.nl/manual/commands.html)  , is very powerful. For instance usually all my file have a header similar to the following snippet : 

```
----------------------------------------------------------------------------
--! @file ff_re.vhd
--! @author el3ctrician
--! @brief Responsible of implementing a flipflop
--! @details a flip flop is a circuit that registers its input on the output only at the rising edge of the clock
--! @version 0.1
--! @date 2020-07-10
--!
--! @copyright 
--!                TerosHDL - GNU Public License
--! is free software: you can redistribute it and/or modify
--! it under the terms of the GNU General Public License as published by
--! the Free Software Foundation, either version 3 of the License, or
--! (at your option) any later version.
```
currently this gets ignores by the documentation tool and nothing is parsed. So adding both the  `@berif` and `@details` commands would be a good start since it is also used sometimes to describe a process. If it used to describe  a process the current behavior is that i found a generated link to github.com/berif or details in my pdf/html. 
We might also consider to support the rest of them and let them be optional in dedicated signals that appears in the output if it is found (version, copyright ..etc.) otherwise it is the same as now.

This could be a great starting point after that we can add a support for grouping signal to one bus as per the following snippet : 

``` vhdl
entity video_core is 
  port (
    --! core clock, 100 Mhz
    clk : in std_logic;
    --! asynchronous active low reset
    reset_n : in std_logic;

   --! @startbus axi_lite_slave 
    s_axi_awaddr  : in std_logic_vector(AXI_LITE_ADDR_WIDTH - 1 downto 0);
    s_axi_awprot  : in std_logic_vector(2 downto 0);
    s_axi_awvalid : in std_logic;
    s_axi_awready : out std_logic;
    s_axi_wdata   : in std_logic_vector(AXI_LITE_DATA_WIDTH - 1 downto 0);
    s_axi_wstrb   : in std_logic_vector((AXI_LITE_DATA_WIDTH/8) - 1 downto 0);
    s_axi_wvalid  : in std_logic;
    s_axi_wready  : out std_logic;
    s_axi_bresp   : out std_logic_vector(1 downto 0);
    s_axi_bvalid  : out std_logic;
    s_axi_bready  : in std_logic;
    s_axi_araddr  : in std_logic_vector(AXI_LITE_ADDR_WIDTH - 1 downto 0);
    s_axi_arprot  : in std_logic_vector(2 downto 0);
    s_axi_arvalid : in std_logic;
    s_axi_arready : out std_logic;
    s_axi_rdata   : out std_logic_vector(AXI_LITE_DATA_WIDTH - 1 downto 0);
    s_axi_rresp   : out std_logic_vector(1 downto 0);
    s_axi_rvalid  : out std_logic;
    s_axi_rready  : in std_logic;
    --! @endbus axi_lite_slave 

    --! @startbus axis_video_in
    video_in_tdata  : in  std_logic_vector(23 downto 0);
    video_in_tlast  : in  std_logic;
    video_in_tuser  : in  std_logic_vector(0 downto 0);
    video_in_tvalid : in  std_logic;
    video_in_tready : out std_logic;
   --! @endbus axis_video_in

    --! @startbus axis_video_out
    video_out_tdata  : out std_logic_vector(23 downto 0);
    video_out_tlast  : out std_logic;
    video_out_tuser  : out std_logic_vector(0 downto 0);
    video_out_tvalid : out std_logic;
    video_out_tready : in  std_logic
   --! @startbus axis_video_out
  );
end video_core;

--! @busdescription axi_lite_slave  An AXI4-Lite compliant slave interface to configure the core registers, the register width is 4 bytes
--! @busdescription axis_video_in An AXI4-Stream compliant slave interface to receive the video frame
--! @busdescription axis_video_in An AXI4-Stream compliant master interface to send the video frame out

``` 

I think that this could improve the readability and the quality of the documentation a lot.

Then maybe, in a very doxygen style, also a way to reference other components in the code and so a global way for documenting the project as all which creates a single pdf/HTML for all the entities and allows to follow hyperlinks between various components which leads to a better documentation for the overall project. A cherry on top at this matter it to support for instance a link to .md file as main page (similar to doxygen ` @mainpage `

Also wavedrom and bitfield are very powerful tools but i would like to see wavedrom supported more than once. This is useful for complex designs where in some cases there are multiple processes inside an entity and so i was thinking maybe we can keep them supported for as once for each process ?

I would love to dedicate some of my spare time to work on some of these features if you can supervise and guide me through the code.

Thanks !

el3ctrician, 2021-01-13:
Hello,
i added a function [here](https://github.com/el3ctrician/colibri/commit/9bb6af9950c7efb77647b4f2a5f84610cebf77b2) to parse the following doxygen commands

- details
- breif
- file
- author
- version
- date
- copyright

the function manipulates the JSON so for the following source
``` vhdl

--! @file axi_interface.vhd
--! @author el3ctrician (elbadriahmad@gmail.com)
--! @brief Some description can be added here
--! also in multi-lines
--! @details Another description can be added here
--! @version 0.1
--! @date 2020-07-10
--! 
--! @copyright Copyright (c) 2019 by HiFuture S.R.L.
--!                HiFuture S.R.L. - PROPRIETARY
--! This file contains proprietary and confidential information.
--! The content or any derivative work can be used only by or distributed to licensed users or owners.
--! Disclosure to third parties or reproduction in any form whatsoever, without prior written consent, is strictly forbidden.
--! 
--!  And more core description can be added here 
entity video_core is
  generic (
    --! number of registers to be read from the AXI Lite interface
    AXI_OUT_REG_NUM       : integer := 10;
    --! number of registers to be written to the AXI Lite interface
    AXI_IN_REG_NUM        : integer := 10;
    --! offset for the start address of the input registers, by default it starts after the read regs
    AXI_IN_REG_START_ADDR : integer := 10
  );
  port (
    --! Global Clock Signal
    S_AXI_ACLK  : in std_logic;
    --! Global Reset Signal. This Signal is Active LOW
    S_AXI_ARESETN  : in std_logic;
    --! Write address (issued by master, acceped by Slave)
    S_AXI_AWADDR  : in std_logic_vector(AXI_LITE_ADDR_WIDTH-1 downto 0);
    --! Write channel Protection type. Note used for this implementation
    S_AXI_AWPROT  : in std_logic_vector(2 downto 0);
    --! Write address valid. This signal indicates that the master signaling valid write address and control information.
    S_AXI_AWVALID  : in std_logic;
    --! Write address ready. This signal indicates that the slave is ready to accept an address and associated control signals.
    S_AXI_AWREADY  : out std_logic;
    --! Write data (issued by master, acceped by Slave)
    S_AXI_WDATA  : in std_logic_vector(AXI_LITE_DATA_WIDTH-1 downto 0);
    --! Write strobes. This signal indicates which byte lanes hold valid data. There is one write strobe bit for each byte of the write data bus.
    S_AXI_WSTRB  : in std_logic_vector((AXI_LITE_DATA_WIDTH/8)-1 downto 0);
    --! Write valid. This signal indicates that valid write data and strobes are available.
    S_AXI_WVALID  : in std_logic;
    --! Write ready. This signal indicates that the slave can accept the write data.
    S_AXI_WREADY  : out std_logic;
    --! Write response. This signal indicates the status of the write transaction.
    S_AXI_BRESP  : out std_logic_vector(1 downto 0);
    --! Write response valid. This signal indicates that the channel is signaling a valid write response.
    S_AXI_BVALID  : out std_logic;
    --! Response ready. This signal indicates that the master can accept a write response.
    S_AXI_BREADY  : in std_logic;
    --! Read address (issued by master, acceped by Slave)
    S_AXI_ARADDR  : in std_logic_vector(AXI_LITE_ADDR_WIDTH-1 downto 0);
    --! Protection type. This signal indicates the privilege and security level of the transaction, not used in this implementation
    S_AXI_ARPROT  : in std_logic_vector(2 downto 0);
    --! Read address valid. This signal indicates that the channel is signaling valid read address and control information.
[... 80 lines cut ...]
``` md

# Entity: video_core
## File
 axi_interface.vhd
## Author
  el3ctrician (elbadriahmad@gmail.com)
## Version
0.1
## Date
2020-07-10
## Copyright
 Copyright (c) 2019 by HiFuture S.R.L.
                HiFuture S.R.L. - PROPRIETARY
 This file contains proprietary and confidential information.
 The content or any derivative work can be used only by or distributed to licensed users or owners.
 Disclosure to third parties or reproduction in any form whatsoever, without prior written consent, is strictly forbidden.
 
## Diagram
![Diagram](example_8.svg "Diagram")
## Description
 Some description can be added here also in multi-lines details Another description can be added here And more core description can be added here 
```

Am not sure about the position of copyrights... maybe it is better to add it to the end of the generated docs

qarlosalberto, 2021-01-13:
Good job @el3ctrician I thint that would be more clear:

# Entity: video_core
- File: axi_interface.vhd
- Author: el3ctrician (elbadriahmad@gmail.com)
- Version: 0.1
- Date: 2020-07-10
- Copyright:
 Copyright (c) 2019 by HiFuture S.R.L.
                HiFuture S.R.L. - PROPRIETARY
 This file contains proprietary and confidential information.
 The content or any derivative work can be used only by or distributed to licensed users or owners.
 Disclosure to third parties or reproduction in any form whatsoever, without prior written consent, is strictly forbidden.
 
## Diagram
[image]
## Description
 Some description can be added here also in multi-lines details Another description can be added here And more core description can be added here

el3ctrician, 2021-01-13:
Ok i'll just keep them bold then :) 

# Entity: video_core

- **File:** axi_interface.vhd
- **Author:** el3ctrician (elbadriahmad@gmail.com)
- **Version:** 0.1
- **Date:** 2020-07-10
- **Copyright:**  Copyright (c) 2021 by TerosHDL
                GNU Public License
    This program is free software: you can redistribute it and/or modify
    it under the terms of the GNU General Public License as published by
    the Free Software Foundation, either version 3 of the License, or
    (at your option) any later version.

    This program is distributed in the hope that it will be useful,
    but WITHOUT ANY WARRANTY; without even the implied warranty of
    MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
    GNU General Public License for more details.

    You should have received a copy of the GNU General Public License
    along with this program.  If not, see <https://www.gnu.org/licenses/>
 
## Diagram
[image]
## Description
 Some description can be added here also in multi-lines details Another description can be added here And more core description can be added here

smgl9, 2021-01-14:
Take a look at https://github.com/TerosTechnology/colibri/projects/2 
I tried to collect the issues and improvements of the documenter in order to organize them. Feel free to comment anything.

el3ctrician, 2021-01-14:
Thanks, currently i can share the following screen shot. I added some test cases and everything seems working fine with the current supported doxygen commands. I am open to any other commands: i was thinking about `@bug` and `@todo` but they are usually not used in a final documentation release. Not sure if it worth it

[image]

smgl9, 2021-01-14:
It looks good :)
The idea is to enable "colibri" as an application eventually. So it could be used in CI to automate the documentation process.  Maybe in that case commands like `@bug` makes more sense. But it's up to you.

closed 2021-03-06 by smgl9