TerosTechnology/colibri #178: mermaid support

smgl9, 2021-01-16:


smgl9, 2021-01-18:
The mermaid block needs to be between keywords. For instance: ```@starmermaid``` ```@endmermaid```

el3ctrician, 2021-01-23:
where should it will be expected ? for each process of for the whole entity ?

qarlosalberto, 2021-01-23:
We could start with the entity description.

el3ctrician, 2021-03-13:
So currently mermaid is parsed and added to the JSON structure, not sure if we can ever add it though since it uses the browser as you can find [here](https://github.com/mermaid-js/mermaid/issues/1183). The cli version [here](https://github.com/mermaid-js/mermaid-cli) uses chromium. I think that in order to this to work without having to pull chromium we could look at [mermaid-vscode](https://github.com/vstirbu/vscode-mermaid-preview) which seems to use the integrated browser in vscode ... this might be a clever hack ....