TerosTechnology/colibri #185: `define paramters confuse the diagram generator.

manub23, 2021-03-04:
Globally declared parameters (using \`define statements) are parsed weirdly in the documentation generator. 

For example, ``input  [`PORT_WIDTH-1:0] port_name;`` is rendered as

[image]

where `` `PORT_WIDTH `` is defined elsewhere as `` `define PORT_WIDTH 32 ``.

It seems as if the markdown parser treats the code in-between two backticks `` ` `` as in-line code.

smgl9, 2021-03-04:
Thank you for reporting it!

We have had very similar issues in the past with some special characters. We wil fix it ;)

manub23, 2021-03-04:
Thanks, @smgl9

smgl9, 2021-03-16:
@manub23 In the last release this error has been fixed. 

Feel free to reopen the issue if you have any problem.

closed 2021-03-16 by smgl9