TerosTechnology/colibri #244: Sphinx builder and autodoc plugin for HDL

GlenNicholls, 2022-01-21:
@umarcor @LarsAsplund @JimLewis @eine @suzizecat @Paebbels I have been talking a bit with @qarlosalberto about potentially creating a sphinx builder/autodoc plugin for HDL, although I'm more interested in VHDL. I would really like the ability to automatically document VHDL code with [autodoc ](https://www.sphinx-doc.org/en/master/usage/extensions/autodoc.html) and a [VHDL language domain](https://www.sphinx-doc.org/en/master/usage/restructuredtext/domains.html) capability. TerosHDL already has [autodoc capability](https://terostechnology.github.io/terosHDLdoc/documenter/configuration.html) for MD/HTML, but it is difficult to manage for large or distributed projects/repos. I am hoping that Sphinx solves this problem and it'd be really nice to eventually have a generic HDL extension in [sphinx-contrib](https://github.com/sphinx-contrib) that anyone can use as long as they have Python and Pip. Plus, I see a lot of potential for this to be very widely used. I decided to ping y'all as I feel that you might find this valuable for your projects and I'm also hoping to get some feedback about how you think this should look. Feel free to ping others. 

Anyways, if you feel that this discussion belongs on Gitter, the VHDL WG issues, or elsewhere, let me know and I can move it. I am planning to talk with management at my company about sponsoring this effort and Carlos was the first person I thought might be interested in taking this on based on Colibri. I previously attempted to go from Colibri to doxygen/asciidoctor. This proved to be a huge headache for distributed projects and is not going to be easy to maintain. I figured reST and Sphinx seemed like a natural choice as there is an interface to create extensions for other languages, builders, etc. The other reason I want to have a discussion about this is there are already quite a few VHDL autodoc plugins for Asciidoctor and others, but most are dead and I don't think any of them really found the right way to document VHDL or present it in the actual HTML/PDF/etc.

# Documenting Source Code

Currently, Colibri and others tend to use some pattern in the comments to differentiate between a comment and documentation. I *believe* colibri can use any symbol, but it defaults to `--!`. I think this is a natural direction because VHDL doesn't have any notion of a docstring or similar. However, I'm not the biggest fan of that pattern because it's extra typing. If your editor/TerosHDL supports carrying on this comment and symbol with ENTER, then it's not that big of a deal. But, if people don't have that then it's annoying to write and update documentation. Using a different pattern in a multi-line comment makes a little bit more sense to me, but that is only valid for 2008+. I don't think this is viable though. 

Instead, I think it should be based on placement similar to Python. In python, you can add a docstring by placing the block comment directly underneath the declaration:

```
SOME_CONST: str = "const"
"""Documentation for SOME_CONST.

Note that I don't think SOME_CONST has a __doc__ attribute and this works based on a sphinx setting or extension
"""

def some_func() -> None:
  """This is built into python, i.e. some_func.__doc__."""
```

For VHDL, I think it makes sense to put the documentation after things as well:

```
entity and_gate is
port (
  a : in std_logic; -- or it could go here
  -- a port documentation

  b : in std_logic;

  -- b port would have no documentation because comment doesn't immediately trail

  z : out std_logic
);
end entity;
-- Documentation for and_gate, ``reST`` is supported and can link to other modules :data:`~lib.pkg.SOME_CONST`.
```

Or even this maybe, adapting from google's style guide for python:

```
entity and_gate is
port (
  a : in std_logic;
  b : in std_logic;

  z : out std_logic
);
end entity;
-- Documentation for and_gate, ``reST`` is supported and can link to other modules :data:`~lib.pkg.SOME_CONST`.
-- Generics:
-- ...
-- Ports:
--     a: a's documentation
--     b:
--        b's documentation
--       
--        .. deprecated:: 1.0 b will be removed in 2.0. 
```

Block comments would also be allowed in the same place. I don't like that the documentation is below the entire entity declaration, maybe it makes sense to adopt Python's notation for classes where the docstring is after the class declaration or something like this maybe?

```
entity and_gate is
-- Documentation for and_gate, ``reST`` is supported and can link to other modules :data:`~lib.pkg.SOME_CONST`.
-- Generics:
-- ...
-- Ports:
--     a: a's documentation
--     b:
--        b's documentation
--       
--        .. deprecated:: 1.0 b will be removed in 2.0. 

port (
  a : in std_logic;
  b : in std_logic;

  z : out std_logic
);
end entity;
```

One unknown here is how to provide documentation for a file. I.e. if you have a license at the top of your file, how does the parser know if there is something below it that should be documentation like https://github.com/VUnit/vunit/blob/master/vunit/__init__.py#L7-L9?

# Layout 

Once the documentation is generated, I HATE the idea that tables are used for ports/generics. I don't think it's intuitive, I feel that tables make it so you focus too much on port attributes like the type or size instead of what the port is and how to use it. Instead, I think that it should just look like it does for python:

[image]

With VHDL, we care more about types/ranges and whatnot, so for this we should include that information in the parameters/generics list.

As for the documentation itself in the image above but for an entity? I think that should be it. All you really care about when instancing it is the API and how to use the component, you don't care about the implementation details and they should be considered private. However, sometimes this is critical to understand, so I think that within the documentation for `and_gate`, there should be a drop-down to expand the architecture (or architectures) for you to view the internal/private documentation like info about signals, FSM's, etc. Teros has some neat features and internal FSM's etc would be cool to reference in an entity's documentation so it shows up when someone scrolls past `and_gate` in the documentation. Another neat idea would be to display the block hierarchy, e.g. `and_gate` and how it connects sub-components. This would be super useful for project documentation where you can go into the top level's documentation and see exactly how everything is connected.

Now for subprograms, packages, etc. it'd just expand on the above ideas to determine a documentation convention along with how it should be displayed.

# Annotations/Autodoc

So for [autodoc](https://www.sphinx-doc.org/en/master/usage/extensions/autodoc.html) with Python in Sphinx, you can do stuff like this for automagically inserting documentation in a reST document:

```
.. automodule:: noodle
   :members:
```

How would these `automodule`, `autoclass`,... look for VHDL/HDL? I don't like the idea that we would specify files directly as I will talk about how we can handle that later. I'm kind of thinking that we could use `autolibrary` which would populate all entitys, packages, architectures, etc in a library:

```
.. autolibrary:: some_lib
    :members: some_pkg, some_ent, some_ent.some_arch ... <- default if this isn't present is to include everything
```

As for other annotations, I think referencing other components, subprograms, etc. would use the library if it exists, otherwise the name itself if it is not in a library:

```
vhdl:ent:`~lib.pkg.ent_name`  <- ~ for only displaying ent_name in docs link
vhdl:func:`lib.pkg.some_function`  <- could be pure or impure, or maybe func and ifunc 
vhdl:proc:`lib.pkg.some_procedure`
vhdl:data:`ent.arch.some_signal`
vhdl:data:`ent.arch.SOME_CONST`
vhdl:type:`lib.pkg.some_type`  <- could be type or subtype, or maybe type and stype
...
```

# Sphinx Building

I am thinking that there would only need to be an HDL extension as the included html builder should still work once it gets the correct information. However, the HDL extension would provide a flag to specify a configuration file. This file would be CSV (or TOML, YAML, or whatever makes sense) that would define all files in the project, their associated library, and anything else needed for building the documentation.

There's probably other stuff I haven't thought about, but this is the gist. Thoughts, suggestions, concerns?

umarcor, 2022-01-21:
My very quick reaction (before going through the details of your message) is:

That's one of @Paebbels' main goals since years. As you can see in [Paebbels/pyVHDLParser: Use Cases](https://github.com/Paebbels/pyVHDLParser#use-cases), he designed the parser the he did because he wanted to preserve all the original content and produce a Document Object Model (DOM) for documentation purposes. Long story short, in Dec 2020 - Jan 2021, he split [vhdl/pyVHDLModel](https://github.com/vhdl/pyVHDLModel) from pyVHDLParser, and then in joint effort with Tristan, they created [pyGHDL.dom](https://github.com/ghdl/ghdl/tree/master/pyGHDL), which is pyVHDLModel + libghdl. In other other words, pyGHDL uses GHDL as a frontend (or backend) to fill the abstraction provided by pyVHDLModel. Note the "Consumers" in [vhdl/pyVHDLModel: Use Cases](https://github.com/vhdl/pyVHDLModel#use-cases).

NOTE: In order to be 100% usable for pyVHDLModel, some enhancements are required in libghdl (GHDL), because by default GHDL e.g. ignores all the comments (which are not needed for simulation or synthesis). Therefore, libghdl needs some "third elaboration mode" which allows to preserve all the details for documentation purposes. Another related enhancement is the resolution of all the symbols. That is all slow work in progress, because pyGHDL.dom is a very exciting use case of GHDL, but not the main one.

pyVHDLModel and pyGHDL.dom are usable since July 2021. A few months ago, I did some proofs of concept to showcase how they can be used. See [Open Source Verification Bundle (OSVB): Project](https://umarcor.github.io/osvb/apis/project.html):

- [OSVB: pyVHDLModelUtils](https://umarcor.github.io/osvb/apis/project/pyVHDLModelUtils.html)
  - Note that it contains functions such as `resolve.Symbols` and `resolve.ArchitecturesToEntities`: https://umarcor.github.io/osvb/apis/project/pyVHDLModelUtils.html#resolve
- [OSVB: Open Source VHDL Design Explorer (OSVDE)](https://umarcor.github.io/osvb/apis/project/OSVDE.html)
- [OSVB: Documentation generation](https://umarcor.github.io/osvb/apis/project/DocGen.html)

As the name suggests, [OSVB: Documentation generation » Integration with Sphinx » Lists and tables](https://umarcor.github.io/osvb/apis/project/DocGen.html#lists-and-tables) shows Lists and Tables for entities, ports, generics, architectures, etc. generated automatically with Python, using pyGHDL.dom and tabulate. Furthermore, subsection [VHDL Domain](https://umarcor.github.io/osvb/apis/project/DocGen.html#vhdl-domain) contains references to "proper" Sphinx extension projects:

- [Paebbels/sphinxcontrib-vhdldomain](https://github.com/Paebbels/sphinxcontrib-vhdldomain)
- [CESNET/sphinx-vhdl](https://github.com/CESNET/sphinx-vhdl)

I suggest to read Paebbels/sphinxcontrib-vhdldomain#4, which is the "latest" discussion with regard to the syntax and directives to use/create. Then, read/check CESNET/sphinx-vhdl. See also the last commit in https://github.com/umarcor/osvb/commits/sphinx-directives, which is somethink I drafted together with Patrick this summer, after chatting with @bradleyharden.

On a different but related topic, see subsection [Diagrams](https://umarcor.github.io/osvb/apis/project/DocGen.html#diagrams). Apart from the references to SymbiFlow's sphinxcontrib-hdl-diagrams (using ghdl + yosys), recently Symbolator and pyHDLParser were forked to org hdl:

- [hdl/symbolator](https://github.com/hdl/symbolator)
- [hdl/pyHDLParser](https://github.com/hdl/pyHDLParser)

There is naturally some overlap between pyHDLParser, pyVHDLModel and pyVHDLParser. We would like to plug pyVHDLModel into Symbolator, in order to decouple it from the parser.

Last, but not least, although we are focused on VHDL, there is also [pySVModel](https://github.com/edaa-org/pySVModel) (see edaa-org/pySVModel#11).

JimLewis, 2022-01-21:
@GlenNicholls   I would be happier with `--!` than being stuck with some automation that tries to find the documentation based on some arbitrary metric.   For example, some might like their entity/subprogram documentation before the entity/subprogram and not after.

umarcor, 2022-01-22:
> it'd be really nice to eventually have a generic HDL extension in sphinx-contrib that anyone can use with pip and sphinx.

Distribution is probably the second most complex challenge here. The first one is knowing/learning how to write proper Sphinx extensions, since the Sphinx codebase/architecture is not the most comfortable to work with. Then, since HDLs are such complex languages, it's not easy to have an extension that works through pip only. In practice, the extension will depend on some compiled/shared library. That is true for VHDL and also for System Verilog, because the most complete open source parsers are written in compiled languages in both cases.

Therefore, most projects trade functionality for distribution complexity. VUnit uses a regexp parser in Python, which is known to be limited, but it allows to not depend on anything other than Python. In fact, we discussed about replacing the parser with rust_hdl in the past, and the main reason not to do so was it being written in Rust, a compiled language. Similarly, TerosHDL uses tree-sitter mainly because it's written in JavaScript, and the current implementation is focused on trying to use as much JS as possible, in order to allow it being installed as "just a VSCode extension".

pyVHDLParser would be a very interesting solution for both VUnit and TerosHDL, because it would require Python only. TerosHDL is mainly written in JS, but it uses VUnit for relevant features, hence the dependency on Python is there already. Unfortunately, pyVHDLParser was started more than a decade later than GHDL, and it's not ready-to-use per se. That's why, a year ago, we decided to split pyVHDLModel from pyVHDLParser and plug it to GHDL (libghdl). We wanted to take the best (most complete) piece from inside pyVHDLParser and make it usable with the best open source VHDL parser/analyzer/elaborator (GHDL).

NOTE: all GHDL, rust_hdl and pyVHDLParser evolved naturally to a very similar internal architecture based on tokens and multiple phases. The main difference between them is the language and how much human-hours were devoted to the development.

The challenge is "how to install the parsers to extract data from sources", which is independent from the type of extension we talk about (documentation as text, entity/component block symbols, post-synthesis block diagrams, GUIs such as hwstudio...). For instance, currently [SymbiFlow/sphinxcontrib-hdl-diagrams](https://github.com/SymbiFlow/sphinxcontrib-hdl-diagrams) supports Verilog only, because it installs Yosys through WASM-pip ([YoWASP](http://yowasp.org/)) or Conda, and Yosys by default does not support System Verilog or VHDL. There is work in progress to support installing the tools through other mechanisms (SymbiFlow/sphinxcontrib-hdl-diagrams#72), but then we have the same issues we find when distributing tools in general (e.g. SymbiFlow/sphinxcontrib-hdl-diagrams#73).

To me, the dependency on GHDL (and Surelog) is perfectly acceptable from a pragmatic point of view, even though the distribution burden is increased. For my use cases, Yosys is not useful without ghdl-yosys-plugin. I need to be able to simulate pre-synthesis model and verify the behaviour, before I deal with post-synthesis results. For that same reason, I would not personally get a meaningful benefit from replacing pyGHDL.dom with pyVHDLParser, or from having GHDL with mcode on YoWASP. It would streamline some containers, but I would need "a regular GHDL" in my stack anyway (because I need to simulate, cosimulate and test the design I'm going to synthesise and document). Nonetheless, I understand that not all the target users of TerosHDL and VUnit are dependent on GHDL, since some of them might use Siemens', Aldec's, Xilinx's or Intel's tools only.

> Plus, I see a lot of potential for this to be very widely used as HDL documentation is an area where I have not seen a solution that meets most people's needs.

- "It's called hardware because it's hard"
- "Software is hard"
- [Parsing VHDL is [very] hard ](https://eli.thegreenplace.net/2009/05/19/parsing-vhdl-is-very-hard)

HDL tooling is blocked since years (decades) because parsing and analysing is very hard, and it's impossible to build tooling without a library that understands the language.
I believe we are all eager to build lots of tools, but all the projects hit the same wall sooner or later. You will find "code generation" to be common in multiple tooling projects, because that's the natural approach when understanding the language is not possible. Something apparently as simple as "getting the direction and type of a port" gets pretty nasty so easily. Actually, it's interesting to try understanding how pyGHDL.dom resolves that.
Hence, we need to support GHDL and Surelog because those are the toolkits that will enable us to provide open source tooling for VHDL and SV on par with "modern" programming languages (Python, Golang, Rust, Scala...).

In parallel, we can and should work on Sphinx directives, because those can be used manually and then plugged to an autodoc. For instance, in GHDL's doc we declare everything manually, because it's an Ada program documented using Sphinx.
The same approach might be used for VHDL projects, until the automation details explained below are solved.

> This proved to be a huge headache for distributed projects and is not going to be easy to maintain.

[... cut ...]
          -> Unless pyVHDLModel is used, a "parser to Python" conversion tool is required.
          -> Python [VHDL semantics]
            -> Python [Sphinx semantics]
```

I suggest we focus on the very last step. How to represent VHDL semantics in terms of Sphinx language domain features.

Note: [pyVHDLModelUtils](https://umarcor.github.io/osvb/apis/project/pyVHDLModelUtils.html) converts pyVHDLModel to restructuredtext, not to Sphinx. I.e., it's not an extension but "Python code executed at build time".

qarlosalberto, 2022-01-22:
I'm away, next week I will do a more detailed comment.

@suzizecat about the external documentation. Is it similar to?: https://terostechnology.github.io/terosHDLdoc/documenter/start.html#special-tags

suzizecat, 2022-01-22:
@qarlosalberto Yay ! (Didn't even know you had those)
However, in Doxygen, you can specify the element you wish to document while the comment is not near said comment as explained in https://stackoverflow.com/questions/18291017/code-documentation-outside-code-file

Therefore you can have the whole documentation separated from the code (including, as example, documentation for internal signals and such)

umarcor, 2022-01-22:
> Support of markdown and Wavedrom

There is a wavedrom extension for Sphinx already. Maybe docstrings can have an optional language keyword, similarly to code blocks in markdown or rst, so that users can specify the markup language used in the docstrings.

> In source documentation

Markdown was designed for single-file documents. Therefore, there is no standard solution to include other files. However, Sphinx and restructuredtext do support `include`, `toctree`, `ref`, intersphinx, etc. With Sphinx, it is trivial to write the content in the sources, in sibling text/rst files, or mixing box approaches (which is the typical solution in Python projects).

For example, fields "Constraints" and "openFPGALoader" in https://hdl.github.io/constraints/Data/Boards/ are generated automatically from YAML sources located in multiple repositories, but shown together (per board). Another example is http://vunit.github.io/examples.html#vhdl, where all the content (the description of each example) is extracted from the docstrings in `run.py` files (note all the references to other sections in the docs).

Paebbels, 2022-01-22:
The doxygen style of comments (`--!`) is not a nice solution. When you look at other languages, they are more clever. E.g. in .NET the double comment character (`//`) is comment and a tipple comment character (`///`) is structured documentation.

I don't suggest to use Markdown as a documentation language, because it's not powerful enough for most purposes. A language like ReStructured Text (ReST) is needed. Maybe it should be a "paragma" or global option how to interpret documentation comments as documentation syntax, because Markdown can't be distinguished from ReST content. Otherwise, the third character after the double comment character could denote the used documentation language.

It's not needed to use different comment indications for *before* and *after* an object. Other languages have managed to define rules so the comment location defined to what object to annotate the comment.

-----------
As @umarcor wrote, there are currently 2 approaches to get comments from VHDL files:
1. pyGHDL.DOM - this needs modifications to GHDL, so comments are preserved and accessible from GHDL's AST. The idea is to preserve tokens and link them to AST nodes. If so, AST nodes can be traversed and then tokens before or after can be found.
2. pyVHDLParser - this is currently on hold. As Unai wrote, an abstract pyVHDLModel was extracted, so multiple frameworks can use that model as a common interface.

Why is it on hold?  
I'm currently working on EDA² and putting all my power into that project to get if flying in the next 3 months. Afterwards, I'll continue either with pyGHDL.DOM if it supports the needed additions or with pyVHDLParser. OTOH, if someone would be interested to help out in pyVHDLParser, I'm open to assist or switch priorities. The parsing technique used in pyVHDLParser is not easy to understand, but very powerful and as it looks in first experiments quite fast.

bradleyharden, 2022-01-23:
First, I want to say that everyone here is much more active and invested in the VHDL ecosystem than I am. So take my opinions for whatever you think they are worth.

That being said, I see **a lot** of what I would consider *very* preliminary bikeshedding going on in this thread.

- Which comment characters are used
- Where the comments are placed
- Whether the system is configurable
- How to configure it

All of these things are trivial details that could be decided and changed later.

In my opinion, there are three **major** pieces that deserve the most attention:

1. How to extract comments from VHDL code
2. How to represent VHDL documentation in reStructured Text and Sphinx
3. How to create links between documented items

There are several potential options for (1). As @umarcor mentioned, distribution of the selected tool could be a challenge. But overall, this area is the most developed of the three and the easiest to get working.

(2) is what I was trying to get at in [Paebbels/sphinxcontrib-vhdldomain#4](https://github.com/Paebbels/sphinxcontrib-vhdldomain/issues/4). I would consider this the biggest challenge. This should be your main focus. (3) can't be handled until (2) is handled anyway.

> The [most complex challenge] is knowing/learning how to write proper Sphinx extensions, since the Sphinx codebase/architecture is not the most comfortable to work with.

I agree with this 100%. Back in 2019, I spent a few weeks working my way through the Sphinx codebase to understand how Python documentation is generated with an eye toward VHDL. I tried to capture some of that in the linked issue. I came away with a few conclusions:
- reStructuredText and Sphinx are not well documented
- The code is medium complexity to understand
- The existing infrastructure within Sphinx is *strongly* linked with the typical notion of software functions. Leveraging this infrastructure to work for VHDL entities would be difficult and awkward.
- The better solution would be a fully custom implementation of a Sphinx domain for VHDL
- However, this would require *even more* knowledge of Sphinx, the reST tree data structure and all of the different passes made over it.

I would **absolutely love** a Sphinx plugin for VHDL. But creating it will be no small project. Before discussing anything else, I would strongly suggest gaining a better understanding of Sphinx and reStructured Text. Otherwise, it will be very easy to accidentally design a reST syntax that ends up being difficult to parse and translate into the correct reST nodes.

In my opinion, everything else in this thread can and should take a back seat to the development of a plan for reST and Sphinx.

Paebbels, 2022-01-23:
According to on of the Sphinx developers, the .NET domain is the most complex domain they did and could server as a starting point for VHDL. Other domains might be too simple.

But as @bradleyharden said, Sphinx might be the main challenge after somehow getting comments from VHDL files. Even with a regexp approach like in VUnit, we would get quite decent results for comment extraction, but have no VHDL domain in Sphinx.

bradleyharden, 2022-01-24:
> According to on of the Sphinx developers, the .NET domain is the most complex domain they did and could server as a starting point for VHDL.

I like this idea overall. My one counter would be that *none* of the software languages may be a good model for VHDL. I would suggest reading the .NET domain implementation while focusing on the broader Sphinx architecture. When I went through the Python implementation, there were a lot of parts that wouldn't really work for VHDL entities. I did learn a bit about how Sphinx nodes are created and recursively processed, but I wish I had understood what to focus on before I started.

suzizecat, 2022-01-24:
My two cents, if one wants to get a hang of Sphinx for HDL, wouldn't be more simple to start from the C++ domain declaration and make changes in order to support SV which is rather close to C++ ? 

This could allow a simpler parser side while helping getting the internal data structure straight, as the internal structure would be closer-ish to VHDL than any other supported (application) language.

bradleyharden, 2022-01-24:
That's an idea. I think there are probably lots of ways to go about it. I was thinking about it a little last night, and the questions that came to mind were:

- What do we want to document to look like?
- Working backwards from there, what reST nodes do you need to produce to get what you want?
- Separately, how should the information be specified in reST text?
- What's left is to connect the dots by parsing the and transforming nodes.

bradleyharden, 2022-02-02:
FYI, someone is advertising the CESNET repo [on Reddit](https://www.reddit.com/r/FPGA/comments/siin1n/sphinxvhdl_automatic_generation_of_documentation/). Maybe it woulw make more sense to start from that project and expand?

umarcor, 2022-02-02:
> FYI, someone is advertising the CESNET repo [on Reddit](https://www.reddit.com/r/FPGA/comments/siin1n/sphinxvhdl_automatic_generation_of_documentation/). Maybe it woulw make more sense to start from that project and expand?

See https://gitter.im/ghdl1/Lobby?at=61336cb85cfd665e5208f098

> I just saw that @lepommosaure started following me, and he has an sphinx-vhdl (https://github.com/lepommosaure/sphinx-vhdl) fork of https://github.com/CESNET/sphinx-vhdl (authored by @jakubcabal and @Adrijaned). I'm pinging the three of them here, to see whether they are interested in talking about pyGHDL and https://github.com/Paebbels/sphinxcontrib-vhdldomain/.